nordicsemi/IOS-nRF-Connect-Device-Manager #21: How to coexist with device manager?

jefffhaynes, 2022-02-10:
Is there a recommended approach for coexisting with the device manager?  I could be mistaken but it seems that (understandably) the device manager transport will assign itself as the peripheral callback delegate.  In the case where our app is looking for its own notifications, it means we lose access to those notifications.  Right now we're simply closing the transport and reassigning the delegate to ourselves, but is there a better way?  Should we be going straight to the transport or possibly doing some delegate multicast trick?

philips77, 2022-02-14:
Hello,
The library should not hijack your connection. Instead, it creates a new `CBCentralManager` instance and retrieves a new instance of a peripheral from it:
https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/blob/f5221e7d1b97c1c8dddd0cae3efe4680a7fe11e4/Source/Bluetooth/McuMgrBleTransport.swift#L113-L126
it should not interfere with your `CBPeripjheral` instance or your delegates.

jefffhaynes, 2022-02-15:
Ah, interesting.  For some reason it seemed like it was interfering but I must have been misinterpreting the symptoms.  I'll keep digging.  Thanks!

closed 2022-02-15 by jefffhaynes