nordicsemi/IOS-nRF-Connect-Device-Manager #23: A few build issues with xcode v13.2.1

MikeSnyder360, 2022-04-10:
I had to do a few extra steps to get this project to almost build. 

FIRST, add the package dependency, THEN do the pod setup. (learning curve) 

$ pod repo update
$ pod install (fails) 
$ pod update SwiftCBOR (works)

in the workspace, the podfiles look good (they were red until I figured out the above steps)

Then when I build in xcode v13.2.1, I get 2 errors 

/Users/mike/work/dor/IOS-nRF-Connect-Device-Manager/Example/Example/View Controllers/Manager/FirmwareUploadViewController.swift:36:48: Cannot convert value of type 'Data' to expected argument type '[ImageManager.Image]' (aka 'Array<(image: Int, data: Data)>')

/Users/mike/work/dor/IOS-nRF-Connect-Device-Manager/Example/Example/View Controllers/Manager/FirmwareUploadViewController.swift:36:32: Incorrect argument label in call (have 'data:delegate:', expected 'images:delegate:')

It's suggesting to replace 'data' with 'images' which also didn't work.

MikeSnyder360, 2022-04-10:
There's a few deprecated things in use here too - 

/Users/mike/work/xxx/IOS-nRF-Connect-Device-Manager/Example/Example/View Controllers/Scanner/ScannerFilterViewController.swift:9:34: Using 'class' keyword to define a class-constrained protocol is deprecated; use 'AnyObject' instead

/Users/mike/work/xxx/IOS-nRF-Connect-Device-Manager/Example/Example/View Controllers/Manager/FirmwareUploadViewController.swift:24:26: 'UIDocumentMenuViewController' was deprecated in iOS 11.0: UIDocumentMenuViewController is deprecated. Use UIDocumentPickerViewController directly.

/Users/mike/work/xxx/IOS-nRF-Connect-Device-Manager/Example/Example/View Controllers/Manager/FirmwareUploadViewController.swift:111:39: 'UIDocumentMenuViewController' was deprecated in iOS 11.0: UIDocumentMenuViewController is deprecated. Use UIDocumentPickerViewController directly.

MikeSnyder360, 2022-04-11:
It seems the upload function wants [Images] which maps to 

`    public typealias Image = (image: Int, data: Data)
`

Right now, its only being provided imageData, which is type Data. 
I changed Line 36 on FirmwareUploadViewController to
`        _ = imageManager.upload(images: [ImageManager.Image(image:0, data:imageData!)], delegate: self)
`
It builds but I cannot imagine this is worth testing with '0' as my image argument. 
It seems to log and pass that image number (0) to the actual uploader, which is used in multiple places.

GriffinJBC, 2022-04-14:
I can confirm the build issues also exist in Xcode 13.3.1. I was able to use the McuManager Framework in my own project by directly importing without issue, however.

I also forked the repo and did a few upgrades to enable support for Carthage, which has also been working well for me.

arthurj58, 2023-03-03:
How to deal with compilation errors
[image]