nordicsemi/IOS-nRF-Connect-Device-Manager #32: Crash in the McuMgrBleTransport class

ramunasjurgilas, 2022-05-17:
using: v1.1.0

Crash screenshot:
[image]

Communication stack and cash reasons:
```
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> -> 0x0000000100480000a0
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> -> 0x0000000100480000a0
debug -> transport -> -> 0x0000001000470000a2666f666673657400656c696d697405
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> -> 0x0000000100480000a0
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff0100001200440001bf6566696c65739fff6673746174757303ff
debug -> transport -> -> 0x0000000100480000a0
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> -> 0x0000001000470000a2666f666673657400656c696d697405
debug -> transport -> -> 0x0000001000440001a2666f666673657400656c696d697405
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> -> 0x0000000100480000a0
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport -> <- 0x0100001700470000bf677265636f7264739fff6969737061727469616cf4ff
debug -> transport iOSMcuManagerLibrary/McuMgrBleTransport.swift:564: Fatal error: Unexpectedly found nil while unwrapping an Optional value
-> -> 0x0000001000470000a2666f666673657400656c696d697405
debug -> transport -> -> 0x0000000100480000a0
2022-05-17 13:35:25.835741+0300 MyHealth[3905:718199] iOSMcuManagerLibrary/McuMgrBleTransport.swift:564: Fatal error: Unexpectedly found nil while unwrapping an Optional value
```

dinesharjani, 2022-05-17:
@ramunasjurgilas there are comprehensive changes in the 1.2 branch which I think fix this. The branch is available here: https://github.com/dinesharjani/mcumgr-ios/tree/1.2-update if you'd like to try it.

ramunasjurgilas, 2022-05-17:
Thanks for update. I will try later.

dinesharjani, 2022-06-02:
@ramunasjurgilas 1.2 was released.

closed 2023-09-07 by dinesharjani