nordicsemi/IOS-nRF-Connect-Device-Manager #47: DFU update: invalidHeaderMagic?

RickGroenewegen, 2022-06-17:
Hi, I'm downloading the firmware using a URL and I try to start the DFU update using that data 

`do {
      try dfuManager.start(data: URLData)
  } catch let error {
      print(error)
  }`

However, this code prints out the following:

`invalidHeaderMagic`

Does anybody have an example on how I should generate the data from an URL that I can pass on to dfuManager.start()?

RickGroenewegen, 2022-06-23:
Figured it out using the example app. I first had to create and pass a package instead of passing the URL:

`do {
    let package = try McuMgrPackage(from: URL(string: ("file://" + fileURL))!)
    try dfuManager.start(images: package.images, using: dfuManagerConfiguration)
} catch {
    print("Error: \(error)")
}`

dinesharjani, 2023-09-07:
So the 'invalid header magic' means the Data is not signed. So, you were maybe trying to DFU Data from a debug build, not Release. This is why it probably worked with your file instead of the Data bytes directly.

closed 2023-09-07 by dinesharjani