nordicsemi/IOS-nRF-Connect-Device-Manager #69: Upgrade successful but upgradeDidComplete does not fire

RickGroenewegen, 2022-08-01:
I'm able to fully upgrade the firmware using this library.

However, upgradeDidComplete does not fire. Instead I'm getting an error:

`2022-08-01 11:39:46.160235+0200 App[4548:292971] [CoreBluetooth] XPC connection invalid`

However, when I comment line 252 in FirmwareUpgradeManager it does work. Line 252 sets the cyclicReferenceHolder to nil:

`cyclicReferenceHolder = nil`

Seems like the reference is removed too quickly or something for upgradeDidComplete to finish?

scsonic, 2022-09-08:
the mcumgr in android will fire Complete
but the ios not, same error here

scsonic, 2022-09-12:
Thansk Rick's informations
finally, my solution is 
```
DispatchQueue.main.async { [weak self] in
    self?.delegate?.upgradeDidComplete()
    self?.cyclicReferenceHolder = nil
}
```

RickGroenewegen, 2022-09-14:
Thanks scsonic! 

I've created a PR: 

https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/pull/72

dinesharjani, 2022-09-19:
PR was merged. Closing for now.

closed 2022-09-19 by dinesharjani