nordicsemi/IOS-nRF-Connect-Device-Manager #87: Variable binding in a condition requires an initializer

RickGroenewegen, 2022-11-09:
Since 1.2.8 a piece of code has been added in McuMgrBleTransport+CBPeripheralDelegate that has been throwing errors. The project cannot be build because of it:

Line 131 : `Variable binding in a condition requires an initializer`

`// Check that we've received all the data for the Sequence Number of the
  // previous recevied Data.
  if let previousUpdateNotificationSequenceNumber,
     !writeState.isChunkComplete(for: previousUpdateNotificationSequenceNumber) {
      
      // Add Data to the previous Sequence Number.
      writeState.received(sequenceNumber: previousUpdateNotificationSequenceNumber, data: data)
      return
  }`

Only option currently is to roll back to <=1.2.7

jazzychad, 2022-12-01:
This is using the new swift syntax for optional unwrapping assignment.. but yes it fails builds for older Xcode/swift versions.. I think I will open a PR with a fix b/c I don't really want to maintain my fork just for this 1 line for much longer :(

jazzychad, 2022-12-01:
made a PR here: https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/pull/98

zhxf2012, 2023-03-07:
I use this framework in my ota sdk which support cocoapods. Unfortunately  I get error below when try to use master:

Analyzing dependencies
[!] Failed to load 'BPBleOTA' podspec: 
[!] Invalid `BPBleOTA.podspec` file: [!] Podspecs cannot specify the source of dependencies. The `:git` option is not supported. `:git` can be used in the Podfile instead to override global dependencies..

 #  from /Volumes/Data/iOSBBB/BPBleOTA/BPBleOTA.podspec:48
 #  -------------------------------------------
 #     s.dependency 'iOSDFULibrary', '~> 4.13.0'
 >     s.dependency 'iOSMcuManagerLibrary' ,:git => 'https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager.git'
 #  end

It seems podspec of cocoapods dosn't support pointing to this master.

Roll back to <=1.2.7 can  work but with an bug leading to ota failed, in my case device skip reset and report completed,but actualy not. Using another mac with new Xcode and latest iOSMcuManagerLibrary it works fine.

Still hope latest version support Xcode 13.x  for a period,since some of my sdk users will still use Xcode 13.x  for a long time.

dinesharjani, 2023-09-07:
Closing for now. Please feel free to reopen if needed.

closed 2023-09-07 by dinesharjani