nordicsemi/IOS-nRF-Connect-Device-Manager #102: Unnecessary broadcast after reboot

vlasty-proglove, 2023-03-15:
Hi all, first, thank you for your contribution and what you have achieved with your SDK.

We have a somewhat specific flow and I noticed that the problem originates from the SDK.
Namely, when uploading firmware to a device with 2 cores and using the option to not reset it after completion, but on the next reboot, the device starts broadcasting for a connection. As for the device with 1 core, the behavior is as expected and does not start broadcasting after a reboot.
Can this automatic broadcast be excluded after a successful install of the firmware on the device with 2 cores as well?

dinesharjani, 2023-09-07:
@vlasty-proglove I don't follow completely. Not reset after completion ? You cannot complete DFU update without a reset. During the reset is when the image swap happens. Then the DFU Library awaits for the device to reset, reconnects, and verifies using the SMP List command that indeed the uploaded images are now running on the two cores.

dinesharjani, 2023-09-13:
I'm referring to @philips77 for this one

philips77, 2023-09-14:
Hello,
Using the `ImageManager` you may send the image without resetting. The `FirmwareUpgradeManager` wraps functionality of `ImageManager` and the `DefaultManager` (should be `OsManager`, and allows automatic DFU using the most common procedure, with 3 modes: *Test+Reset*, *Confirm+Reset*, *Test+Reset+Confirm*, but when you only want to upload the image and send *Test* command, use the `ImageManager` directly.

In nRF Connect Device Manager app you may do this using the Advanced pane of the Image tab.

vlasty-proglove, 2023-09-19:
Thank you @philips77 and @dinesharjani I found a solution that might benefit me.
Yes, as you said, I need to use the ImageManager directly, the "list" method returns pending images, so I am able to do something with that.

vlasty-proglove, 2023-10-11:
Thank you for your replies, I resolved my problem and will close the ticket.

closed 2023-10-11 by vlasty-proglove