nordicsemi/IOS-nRF-Connect-Device-Manager #182: Handle multiple requests one after each other

mpiffari, 2023-12-06:
Hello folks and @dinesharjani  - @philips77 - @roshanrajaratnam,
I'm struggling with the possibility of sending multiple requests one after each other without waiting for the response, and then receiving responses with matching sequence numbers.
Do you have more information about this? Is it still possibile, in order to speed up procedure?

mpiffari, 2023-12-11:
Hi @philips77 ,
thanks for your comment.

Just two questions:
- How can I obtain such an high MTU? With a Write (and not a write without response)? Or I need to set some specific flag in config file on firmware side?
- DLE: is a feature that need to be enabled via config file?

philips77, 2024-03-14:
Hello,
Sorry for the delay, I missed your questions.

> How can I obtain such a high MTU? With a Write (and not a write without response)? Or I need to set some specific flag in config file on firmware side? (maybe CONFIG_NCS_SAMPLE_MCUMGR_BT_OTA_DFU_SPEEDUP?)

In nRF Connect SDK you need to set a KConfig option to set the MTU. This will allow sending Write Without Response packets of large size on the ATT layer.

For Write With Response the packet size is always at most 512 bytes, but this is sent using Prepare Write commands if MTU is not high enough. For uploading data it's better to use Write Without Response, just negotiate the MTU to be 498.

> DLE: is a feature that need to be enabled via config file?

Yes, this is also enabled in the KConfig file. I think you use `CONFIG_BT_CTLR_DATA_LENGTH_MAX` for that, with max value of 251. This defines the max size of a packet on L2CAP layer. It this length is not enough to send large packet (MTU>DLE), then the ATT layer will automatically split longer packets.

dinesharjani, 2025-08-13:
Closing since I think @philips77 provided resolution. Feel free to reopen if needed.

closed 2025-08-13 by dinesharjani