nordicsemi/IOS-nRF-Connect-Device-Manager #198: [Question] How to tell what's the integer error-code when an upload fails?

ksidirop-laerdal, 2024-02-08:
I'm new to swift so please bare with me. I noticed that in McuManager.swift we have this piece of code:

```
extension McuMgrReturnCode: CustomStringConvertible {
    
    public var description: String {
        switch self {
        case .ok:
            return "OK (0)"
        case .unknown:
            return "Unknown (1)"
        case .noMemory:
            return "No Memory (2)"
        case .inValue:
            return "In Value (3)"
        case .timeout:
            return "Timeout (4)"
        case .noEntry:
            return "No Entry (5)"
        case .badState:
            return "Bad State (1)"
        default:
            return "Unrecognized (\(rawValue))"
        }
    }
}
```

When an upload fails the following callback gets triggered:

```
public func uploadDidFail(with error: Error) {
     //  ...
}
```

How can I get the integer error-code (McuMgrReturnCode) from the 'error' parameter? Is it even possible or is it a lost cause? Appreciate any insights.

dinesharjani, 2024-03-12:
You can typecast the error.

So something like 

`if let mcuMgrError = error as? McuMgrError {
   // here mcuMgrError is of type McuMgrError which can be either case returnCode or case groupCode
}`

closed 2024-03-12 by dinesharjani

ksidirop-laerdal, 2024-11-04:
For anyone that stumbles upon this in the future this is how I ended up deducing the numeric error codes:

```swift
        guard let mcuMgrError = error as? McuMgrError else {
            return -99
        }

        var errorCode = -99
        var groupErrorCode = -99
        var groupSubsystemId = 0

        switch mcuMgrError {
        case .returnCode(let rc): //if smp v2 is not supported by the target device (rare these days)
            errorCode = Int(rc.rawValue)
        case .groupCode(let groupCode): //if smp v2 is supported by the target device
            groupErrorCode = Int(groupCode.rc.rawValue)
            groupSubsystemId = Int(groupCode.group)
        }
```

(swift is not my realm of expertise so take this with a grain of salt)