nordicsemi/IOS-nRF-Connect-Device-Manager #202: Readme instructions for "Multi-Image DFU Format"

RaemondAt15, 2024-03-06:
I just wanted to note that the Multi-Image DFU Format example doesn't work without adding the example's Util/McuMgrPackage.swift and Util/UTI.swift as well as the ZIPFoundation pod. It would be useful to mention this or include them in the package.

Thanks,
Raemond

dinesharjani, 2024-03-12:
You mean the Example app doesn't build? I don't follow completely

dinesharjani, 2024-03-14:
Added the `import` statements to the Readme file's code examples. ZIP Foundation is needed as you mentioned, but via SPM it should be pulled auto-magically for you.

closed 2024-03-14 by dinesharjani

RaemondAt15, 2024-03-14:
Sorry. I don't mean the example doesn't build. What I meant is if you are just importing the iOS package to your own project and trying to follow the readme instructions for multi-image dfu, it doesn't work. unless you also copy over those util libraries. My point is that the readme's example usage should work without having to figure out what util libraries have to be copied over.

dinesharjani, 2024-03-15:
Are you using Cocoapods or Swift Package Manager?