nordicsemi/IOS-nRF-Connect-Device-Manager #215: iOS binding from Xamarin, incomplete iOSMcuManagerLibrary.framework headers

lele4warp, 2024-06-11:
I'm trying to import the native libraries to Xamarin forms, I need for the FirmwareUpgrade section of the  iOSMcuManagerLibrary.
I compile the example project correctly and I can access to the iOSMcuManagerLibrary.framework, but seems that the .h is incomplete, infact starting the sharpie process the API definition.cs doesn't contains the fw update functions.
Can you please check that the .h contains all the needed interfaces? If not the binding process cannot generate the fw update functions that I need

boris-sclauzero-zupit, 2024-06-12:
I am working with @lele4warp 
This is the file generated by sharpie command line tool.
I can find only the definition for the object: `McuMgrBleTransport`
We suppose there is something missing in the file `iOSMcuManagerLibrary-Swift.h`

```cs
using CoreBluetooth;
using Foundation;

namespace iOSMcuManagerLibrary
{
	// @interface McuMgrBleTransport : NSObject
	[BaseType (typeof(NSObject), Name = "_TtC20iOSMcuManagerLibrary18McuMgrBleTransport")]
	[DisableDefaultCtor]
	interface McuMgrBleTransport
	{
	}

	// @interface iOSMcuManagerLibrary_Swift_313 (McuMgrBleTransport) <CBPeripheralDelegate>
	[Category]
	[BaseType (typeof(McuMgrBleTransport))]
	interface McuMgrBleTransport_iOSMcuManagerLibrary_Swift_313 : ICBPeripheralDelegate
	{
		// -(void)peripheral:(CBPeripheral * _Nonnull)peripheral didDiscoverServices:(NSError * _Nullable)error;
		[Export ("peripheral:didDiscoverServices:")]
		void Peripheral (CBPeripheral peripheral, [NullAllowed] NSError error);

		// -(void)peripheral:(CBPeripheral * _Nonnull)peripheral didDiscoverCharacteristicsForService:(CBService * _Nonnull)service error:(NSError * _Nullable)error;
		[Export ("peripheral:didDiscoverCharacteristicsForService:error:")]
		void Peripheral (CBPeripheral peripheral, CBService service, [NullAllowed] NSError error);

		// -(void)peripheral:(CBPeripheral * _Nonnull)peripheral didUpdateNotificationStateForCharacteristic:(CBCharacteristic * _Nonnull)characteristic error:(NSError * _Nullable)error;
		[Export ("peripheral:didUpdateNotificationStateForCharacteristic:error:")]
		void Peripheral (CBPeripheral peripheral, CBCharacteristic characteristic, [NullAllowed] NSError error);

		// -(void)peripheral:(CBPeripheral * _Nonnull)peripheral didUpdateValueForCharacteristic:(CBCharacteristic * _Nonnull)characteristic error:(NSError * _Nullable)error;
		[Export ("peripheral:didUpdateValueForCharacteristic:error:")]
		void Peripheral (CBPeripheral peripheral, CBCharacteristic characteristic, [NullAllowed] NSError error);
	}

	// @interface iOSMcuManagerLibrary_Swift_322 (McuMgrBleTransport) <CBCentralManagerDelegate>
	[Category]
	[BaseType (typeof(McuMgrBleTransport))]
	interface McuMgrBleTransport_iOSMcuManagerLibrary_Swift_322 : ICBCentralManagerDelegate
	{
		// -(void)centralManagerDidUpdateState:(CBCentralManager * _Nonnull)central;
		[Export ("centralManagerDidUpdateState:")]
		void CentralManagerDidUpdateState (CBCentralManager central);

		// -(void)centralManager:(CBCentralManager * _Nonnull)central didConnectPeripheral:(CBPeripheral * _Nonnull)peripheral;
		[Export ("centralManager:didConnectPeripheral:")]
		void CentralManager (CBCentralManager central, CBPeripheral peripheral);

		// -(void)centralManager:(CBCentralManager * _Nonnull)central didDisconnectPeripheral:(CBPeripheral * _Nonnull)peripheral error:(NSError * _Nullable)error;
		[Export ("centralManager:didDisconnectPeripheral:error:")]
		void CentralManager (CBCentralManager central, CBPeripheral peripheral, [NullAllowed] NSError error);

		// -(void)centralManager:(CBCentralManager * _Nonnull)central didFailToConnectPeripheral:(CBPeripheral * _Nonnull)peripheral error:(NSError * _Nullable)error;
		[Export ("centralManager:didFailToConnectPeripheral:error:")]
		void CentralManager (CBCentralManager central, CBPeripheral peripheral, [NullAllowed] NSError error);
	}

	[Static]
	[Verify (ConstantsInterfaceAssociation)]
	partial interface Constants
	{
		// extern double iOSMcuManagerLibraryVersionNumber;
		[Field ("iOSMcuManagerLibraryVersionNumber", "__Internal")]
		double iOSMcuManagerLibraryVersionNumber { get; }

		// extern const unsigned char[] iOSMcuManagerLibraryVersionString;
		[Field ("iOSMcuManagerLibraryVersionString", "__Internal")]
		byte[] iOSMcuManagerLibraryVersionString { get; }
	}
}
```

philips77, 2024-06-13:
Try cloning the repo and modify the classes you need, for example `McuManager` with `NSObject`, like the `McuMgrBleTransport` has.

```swift
open class McuManager : NSObject {
   // [...]
}
```

dinesharjani, 2025-09-22:
Closing since there hasn't been any feedback after https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/pull/216

closed 2025-09-22 by dinesharjani