nordicsemi/IOS-nRF-Connect-Device-Manager #219: Insufficient MTU: 253.

RoeeNadam, 2024-06-16:
What can be the reason that I got this 253? 
The write is successful but at a low rate of 0.3-0.4 kb/sec.

dinesharjani, 2024-07-31:
Hi @RoeeNadam - this is "standard behaviour" in a way.

Usually it means you set the MTU higher than is supported by the current device / connection, so you need to set it to the value given with the error (it's an enum with an associated value, if memory serves right). You can check how this is handled in ImageManager: https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/blob/571c42e2a84bc897233e0e1ad892c4c1b8f7caee/Source/Managers/ImageManager.swift#L388

everuribe, 2024-08-21:
I'm also seeing this error:
`Request (SMPv2, group: image, seq: 120, command: upload) failed: Insufficient MTU: 505.`

But our MTU is set to 512...
```
# RX/TX packet sizes
CONFIG_BT_BUF_ACL_RX_SIZE=512
CONFIG_BT_L2CAP_TX_MTU=512
```

dinesharjani, 2024-08-22:
@everuribe the error is in the logs, or DFU fails with this error? The error message might show up in the logs to "correct" the MTU down to the maximum length. As for that seven byte disparity, it could be many things. The McuMgrHeader is 7-8 bytes I think, could account for that, but don't hold me to it. Important thing: does DFU outright fail, or is it that you see this message on the logs? The latter is OK and expected behaviour.

dinesharjani, 2025-03-20:
I still don't know if MTU doesn't work for you. I assume it does, or you would've said so otherwise.

Basically, this "error" is received usually on the first BLE request sent to the firmware, so that the library can self-regulate the proper MTU size of the current communication channel. We assume what the underlying CoreBluetooth library's maximum MTU to begin with, but usually after the first request is sent we correct it. Usually to lower it.

dinesharjani, 2025-07-21:
Closing issue due to lack of follow-up.