nordicsemi/IOS-nRF-Connect-Device-Manager #251: Incompatible dependencies - ZIPFoundation

nysander, 2024-07-15:
```
[!] CocoaPods could not find compatible versions for pod "ZIPFoundation":
  In snapshot (Podfile.lock):
    ZIPFoundation (= 0.9.19)

  In Podfile:
    iOSDFULibrary (= 4.15.3) was resolved to 4.15.3, which depends on
      ZIPFoundation (= 0.9.19)

    iOSMcuManagerLibrary (= 1.7.0) was resolved to 1.7, which depends on
      ZIPFoundation (= 0.9.11)
      ```

MalcolmnDEV, 2024-07-15:
[image]
I get a similar issue using SPM

closed 2024-07-31 by dinesharjani

nysander, 2024-07-31:
what about cocoapods?

dinesharjani, 2024-07-31:
hmm, can you make a pull request for cocoapods ?

dinesharjani, 2024-07-31:
I'll try to do it. Fingers crossed it's a simple version bump

dinesharjani, 2024-07-31:
@nysander we'd have to raise minimum deployment version to 12.0, looks like.

nysander, 2024-07-31:
for us it's not a problem we support 14.0+ but we use your dfu and smp libraries in paralel for different devices so having dependencies aligned is important for us.

thanks for update in advance

dinesharjani, 2024-07-31:
Alright. I'll make a PR with a branch for you to test it out.

dinesharjani, 2024-07-31:
Pull Request: https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/pull/254

nysander, 2024-07-31:
pod update and build successful

dinesharjani, 2024-07-31:
Merged

PR #252 (2024-07-31): Resolve missing package: ZIPFoundation
This PR attempts to resolve a missing package from release 1.7

Close: #251