nordicsemi/IOS-nRF-Connect-Device-Manager #337: [1.9.2] FirmwareUpgradeManager.start() fails with an error about the 'mtu' value even though it used to work just fine with 1.9.0

ksidirop-laerdal, 2025-03-17:
Eversince we upgraded Laerdal.McuMgr C# nugets to use version 1.9.1/1.9.2 we noticed that the firmware installation started failing upon attempting to perform the firmware-upload. The error we're seeing in the logs is this one (trimmed for brievity):

```
Crashed Thread:        0  tid_103  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x0000000106513a28

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [7891]

Thread 0 Crashed:: tid_103 Dispatch queue: com.apple.main-thread
0   McuMgrBindingsiOS             	       0x106513a28 Swift runtime failure: Unexpectedly found nil while implicitly unwrapping an Optional value + 0 [inlined]
1   McuMgrBindingsiOS             	       0x106513a28 FileSystemManager.upload(name:data:offset:callback:) + 1044 (FileSystemManager.swift:70)
2   McuMgrBindingsiOS             	       0x106514354 FileSystemManager.upload(name:data:using:delegate:) + 1328 (FileSystemManager.swift:182)
3   McuMgrBindingsiOS             	       0x1064cd548 IOSFileUploader.beginUpload(_:_:_:_:) + 2160 (IOSFileUploader.swift:128)
4   McuMgrBindingsiOS             	       0x1064cd708 @objc IOSFileUploader.beginUpload(_:_:_:_:) + 144
5   Laerdal.McuMgr.Testbed.Maui   	       0x105167194 xamarin_dyn_objc_msgSend + 160 (trampolines-arm64-objc_msgSend.inc.s:99)
6   Laerdal.McuMgr.Testbed.Maui   	       0x105503414 do_icall + 368 (interp.c:2322)
7   Laerdal.McuMgr.Testbed.Maui   	       0x105501980 do_icall_wrapper + 356 (interp.c:2350)
8   Laerdal.McuMgr.Testbed.Maui   	       0x1054f7288 mono_interp_exec_method + 2948
9   Laerdal.McuMgr.Testbed.Maui   	       0x1054f4e28 interp_runtime_invoke + 244 (interp.c:2109)
10  Laerdal.McuMgr.Testbed.Maui   	       0x105442fc8 mono_jit_runtime_invoke + 1320 (mini-runtime.c:3683)
11  Laerdal.McuMgr.Testbed.Maui   	       0x1053e5288 do_runtime_invoke + 60 (object.c:2576) [inlined]
12  Laerdal.McuMgr.Testbed.Maui   	       0x1053e5288 mono_runtime_try_invoke + 156 (object.c:2733)
```

We managed to narrow down the issue on this commit:

[image]

This assumes that the .mtu property has been properly set via the constructor:

[image]

But for some reason the call to:

```swift
centralManager.retrievePeripherals(withIdentifiers: [targetIdentifier])
```

doesn't seem to detect the peripheral. In our case we use this constructor:

```
_transporter = McuMgrBleTransport(cbPeripheral)
```

https://github.com/Laerdal/Laerdal.McuMgr/blob/3d78886da96a6396a9dcfb3b13744ae565c8e964/Laerdal.McuMgr.Bindings.MacCatalystAndIos.Native/McuMgrBindingsiOS/McuMgrBindingsiOS/IOSFirmwareInstaller.swift#L19 

Which in turn calls the guid constructor internally that tried to re-discover the peripheral using the same guid causing the problem at hand (at least that's our theory)

Quick idea, off of the top of my head:

Can the 'McuMgrBleTransport' constructor be simplified to be completely self-reliant and thus solve the issue at hand (while also being more bulletproof)?

```
public init(_ target: CBPeripheral) {
        
        self.centralManager = CBCentralManager(delegate: nil, queue: .global(qos: .userInitiated))
        self.identifier = target.identifier
        self.connectionLock = ResultLock(isOpen: false)
        self.writeState = McuMgrBleTransportWriteState()
        self.observers = []
        self.operationQueue = OperationQueue()
        self.operationQueue.qualityOfService = .userInitiated
        self.operationQueue.maxConcurrentOperationCount = 1
        
        super.init()
        
        self.centralManager.delegate = self
        self.peripheral = target
        self.mtu = min(target.maximumWriteValueLength(for: .withoutResponse), McuManager.getDefaultMtu(scheme: getScheme()))
}
```

PS: We've also noticed the following piece of logic in the original crash:

```swift
let maxPacketSize = max(maxReassemblySize, UInt64(transport.mtu))
```

Would it make more sense to use 'min' instead of 'max' considering that the 'maxReassemblySize' is the upper bound?

```swift
let maxPacketSize = min(maxReassemblySize, UInt64(transport.mtu))
```

( Or is it that the 'maxReassemblySize' has been mis-named and it should have been called 'minReassemblySize' instead? )

ksidirop-laerdal, 2025-03-20:
PS: Added a PR about this

https://github.com/NordicSemiconductor/IOS-nRF-Connect-Device-Manager/pull/338

ksidirop-laerdal, 2025-04-01:
Any update on this?

dinesharjani, 2025-07-28:
Closing since the proposed PR fix was merged.

closed 2025-07-28 by dinesharjani