nordicsemi/IOS-nRF-Connect-Device-Manager #361: [1.9.2] Firmware Installer: The default mtu of the transporter (still) comes pre-baked as just 20 causing the firmware installation to stall and fail very early and very slowly

ksidirop-laerdal, 2025-07-29:
If we set the mtu value manually to something like '250' everything works just fine so we have a stop-gap solution for now - but still ...

Our code is fairly straightforward:

```
_transporter = _transporter == nil
                ? McuMgrBleTransport(_cbPeripheral)
                : _transporter

logMessageAdvertisement("[IOSFI.ETIIEO.020] using pre-set initial-mtu-size transporter.mtu='\(String(describing: _transporter.mtu))'", McuMgrLogCategory.transport.rawValue, McuMgrLogLevel.info.name)
```

I would expect the mtu to be derived from:

```
cbPeripheral.maximumWriteValueLength(for: .withoutResponse)
```

But it always prints '20' for some reason.

dinesharjani, 2025-07-30:
"Firmware Installer" is FileSystemManager?

ksidirop-laerdal, 2025-07-30:
> "Firmware Installer" is FileSystemManager?

I'm talking about FirmwareUpgradeManager (updated the code sample to reflect this)

dinesharjani, 2025-08-05:
Hey @ksidirop-laerdal I made some changes in main. Could you update the library to main and try again? You should not need to set the mtu yourself anymore.

ksidirop-laerdal, 2025-08-05:
Using latest commit a30fc0b6d205d86e07a7f86f076064998131c94f at the time of this writing from the main branch of ios-mcumgr.

Still seeing mtu=20 when I allow it to auto-configure itself:

[image]

Our approach is pretty straightforward as you can see in the source-code:

https://github.com/Laerdal/Laerdal.McuMgr/blob/develop/Laerdal.McuMgr.Bindings.MacCatalystAndIos.Native/McuMgrBindingsiOS/McuMgrBindingsiOS/IOSFirmwareInstaller.swift

dinesharjani, 2025-08-08:
@ksidirop-laerdal I'm trying to make sense of your code and why the default McuMgrBleTransport's MTU size would be set to 20. I'm going to patch it to enforce it to a higher number so at least it doesn't fail due to the size of the header, but, reasoning about the code the only answer I can come up with right now for the value to be set to 20 would be in this line of code 

`peripheral!.maximumWriteValueLength(for: .withoutResponse)`

So my question is, this happens I imagine with a specific firmware you're trying to update with, right? If you use one of our out-of-the box samples, you still get MTU 20? Because, as far as I can think of right now, the 20 bytes MTU has to be derived from the initial connection parameters. I'm going to try and play around with some firmware build settings but, I'm just asking in case something was modified in the firmware parameters to cause the value of 20 to be returned or, if you can point to a sample, like peripheral_lbs from the NCS SDK or smp_svr sample, etc. that you can say "made a new app from this sample, and MTU is 20 on [x] model of iPhone running [x] version of iOS".

ksidirop-laerdal, 2025-08-08:
Appreciate the prompt analysis on this issue. I will experiment with the bleeding-edge and let you know how it goes.

ksidirop-laerdal, 2025-08-08:
Good news:

The auto-configure mode starts the firmware installation with an mtu=73 on my Apple M1 Laptop (MacCatalyst).

Fair enough even though it doesn't use the max-mtu=450.

dinesharjani, 2025-08-11:
@ksidirop-laerdal uhm, I think when I first read this I understood something different - by "auto-configure" you mean the default MTU value that's returned after McuMgrBleTransport's init? When before it'd return "20" it now returns "73"? Or you're referring to a firmware build setting?

ksidirop-laerdal, 2025-08-12:
> "20" it now returns "73"?

Yes exactly. This happens with the latest bleeding edge of the main branch of these swift-libs. It's better than before even though not ideal (ideal values would have been around mtu=450+ based on our experiments).

The way we log the mtu value can be reviewed here:

https://github.com/Laerdal/Laerdal.McuMgr/blob/develop/Laerdal.McuMgr.Bindings.MacCatalystAndIos.Native/McuMgrBindingsiOS/McuMgrBindingsiOS/IOSFirmwareInstaller.swift#L150

Very simple really.

dinesharjani, 2025-08-12:
The issue is that I canno treplicate the issue. This is why I asked for the KConfig values you're setting on your specific SMP Server Implementation, or for a .hex file I can flash to test. There are other options, like a BLE Sniffer to see the BLE Connection Parameters exchange. But anyway, you can try to set the MTU to the new `SMP_SVR_BT_L2CAP_MTU`, or set it to the value of 495, and if that works for you and is better, we can consider the issue closed.

ksidirop-laerdal, 2025-08-12:
> The issue is that I canno treplicate the issue. This is why I asked for the KConfig values you're setting on your specific SMP Server Implementation, or for a .hex file I can flash to test. There [...]

Give me a few days and I'll get back to you on this one.

dinesharjani, 2025-08-13:
@ksidirop-laerdal I will be Out of Office until September 1.

ksidirop-laerdal, 2025-08-13:
We're gonna put together a testbed and share it with you at some point - might take some time so we can't give you a definitive date. We'll keep you posted when we have something ready.

closed 2025-10-24 by dinesharjani