nordicsemi/IOS-nRF-Connect-Device-Manager #376: Crash when calling fsManager.download() - unrecognized selector error

matthewfx, 2025-08-13:
## Description
When attempting to download a file using the FileSystemManager, the application crashes with an "unrecognized selector" error related to `McuMgrBleTransport.centralManagerDidUpdateState:`.

## Environment
- **Device**: iPhone 16 Pro Max
- **iOS Version**: 18.5
- **Target Hardware**: nRF5340 Development Kit
- **Library Version**: 1.9.3
- **Xcode Version**: 16.4

## Code Sample
```swift
let bleTransport = McuMgrBleTransport(cbPeripheral)
bleTransport.logDelegate = self
let fsManager = FileSystemManager(transport: bleTransport)

let downloadStarted = fsManager.download(name: fullFilePath, delegate: self)
```

## Error Message
```
[iOSMcuManagerLibrary.McuMgrBleTransport centralManagerDidUpdateState:]: unrecognized selector sent to instance 0x1556f0b40
```

## Expected Behavior
The file download should initiate successfully and the `FileDownloadDelegate` should receive appropriate callbacks for download progress and completion.

## Actual Behavior
Application crashes immediately when `fsManager.download()` is called with the unrecognized selector error.

## Additional Context
- I have successfully implemented a basic file download mechanism that works, but it lacks proper packet verification, error handling, and connection drop recovery
- The crash appears to be related to Core Bluetooth state management within the transport layer
- The `cbPeripheral` object is valid and connected before creating the transport

## Workaround
Currently using a custom implementation without the library's robust error handling and verification features.

## Investigation Needed
This appears to be a potential issue with:
1. Core Bluetooth delegate method implementation in `McuMgrBleTransport`
2. Possible iOS 18 compatibility issue
3. Missing delegate setup or protocol conformance

dinesharjani, 2025-08-13:
Hey @matthewfx - have you checked if the current in-development version of the library also crashes? Can you target main branch instead of the last release and test?

Also, can you share full logs until the crash from the library side?

dinesharjani, 2025-08-13:
I have tried to build SMP_SVR for 5340DK but unfortunately I can't get it to advertise over BLE. But for 52840, I tried to connect to a device, and then trigger a Download and I can't replicate the crash. It's also interesting because McuMgrBleTransport does definitely implement the "unrecognized selector", centralManagerDidUpdateState. Hmmm, I guess it could be related to AccessorySetupKit then? I'll try to look into that.

matthewfx, 2025-08-14:
Hey @dinesharjani 
Thanks for the quick response! Here are my findings:

## Testing Status
I haven't tested the current in-development version yet - I'll switch to the main branch and test that next.

## Logging Issues
The log function `func log(_ msg: String, ofCategory category: iOSMcuManagerLibrary.McuMgrLogCategory, atLevel level: iOSMcuManagerLibrary.McuMgrLogLevel)` is never called, so I'm not getting any library-side logs leading up to the crash.

## Key Finding: Peripheral is nil
I discovered what might be the root cause. I added breakpoints at two locations:
1. Right after calling `McuMgrBleTransport.init(peripheral)`
2. Just before calling `fsManager.download()`

In both cases, the `McuMgrBleTransport.peripheral` property was `nil`. This could explain why `centralManagerDidUpdateState:` isn't being recognized - the transport may not be properly initialized with the peripheral.

## Troubleshooting Attempts
1. **AccessorySetupKit**: Tried invalidating the AccessorySetupKit session before creating the transport - no difference
2. **CBCentralManager conflict**: Attempted disconnecting the peripheral from my CBCentralManager instance before passing it to `McuMgrBleTransport(peripheral)`, thinking there might be a conflict with multiple central manager instances - this didn't help either

## Questions
- Is there a specific peripheral state or connection requirement before creating the McuMgrBleTransport?
- Should the peripheral be disconnected from the app's CBCentralManager before being passed to the library?
- Are there any known issues with peripheral retention in the transport layer?

I'll test with the main branch next and report back. Let me know if you need any additional debugging information!

matthewfx, 2025-08-14:
## Update: Main Branch Testing Results

Just tested the main branch with interesting findings:

### Fixed Issue
✅ **Peripheral retention**: The peripheral is now properly retrieved and no longer nil (without any changes to my code)

### Remaining Issues  
❌ **Same crash**: App still crashes with the identical error:
```
[iOSMcuManagerLibrary.McuMgrBleTransport centralManagerDidUpdateState:]: unrecognized selector sent to instance 0x....
```

❌ **No library logs**: Still not receiving any library-side logs through the log delegate

matthewfx, 2025-08-15:
Hi @dinesharjani I understand. I will probably stick with my own implementation for now because we have some demos and I need something to work. Please let me know when you're back. Have a great holiday!

matthewfx, 2025-08-19:
@dinesharjani Yes.

matthewfx, 2025-08-29:
## RESOLVED: Root Cause Identified - Tuist Linker Configuration

@dinesharjani After extensive investigation, I've identified the root cause of this issue.

### Investigation Process
1. **Example app testing**: Your example app worked perfectly
2. **AccessorySetupKit integration**: Added AccessorySetupKit to the example app - still worked fine
3. **Linker analysis**: The error suggesting missing delegate implementation led me to investigate linking issues

### Root Cause
The crash was caused by **incomplete linking in Tuist-based projects**. Our app uses Tuist for modularization, and while we haven't encountered issues with other external dependencies, your package requires the `-ObjC` linker flag to be properly linked.

### Solution
Adding the `-ObjC` flag to the Tuist project configuration resolved the crash completely.

### Technical Context
The `-ObjC` flag is required because your library likely contains Objective-C categories or runtime features that aren't automatically included by the linker without this flag. This is a common requirement for libraries with mixed Swift/Objective-C code or Core Bluetooth implementations.

### Recommendations
Consider either:
1. **Library modification**: Restructure the package to avoid requiring the `-ObjC` flag (if feasible)
2. **Documentation**: Add a note in README.md about the `-ObjC` requirement for Tuist projects and potentially other build systems

This issue demonstrates the importance of testing with various build configurations and tools beyond standard Xcode projects.

Thank you for your help!

closed 2025-08-29 by matthewfx